robotpy/pyfrc #11: Deploying sometimes fails?

aerobit, 2015-01-08:
For some reason, the deploy command doesn't seem to work on a specific windows 8 pyfrc install I'm using.

This line right here https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/mains/cli_deploy.py#L102
only works when it's changed to ` controller.sftp(tmp_dir, deploy_dir)`

This is weird, and might be a side effect of different dev tools on windows?

It also might be localized to that machine; I'll do further testing if needed.

virtuald, 2015-01-08:
I suspect it is a Windows difference, I've done most of my development on OSX/Linux. I'll give it a look later tonight.

aerobit, 2015-01-08:
I think given a folder, psftp will just copy the file without the folder; haven't actually tested it though