robotpy/pyfrc #15: netconsole-host only sends stderr via netconsole port

virtuald, 2015-01-11:
Quite unfortunate, really.

PeterJohnson, 2015-01-11:
Java and C++ stdout go through netconsole.. what's different about Python?

virtuald, 2015-01-11:
That's an excellent question.

PeterJohnson, 2015-01-11:
I just tested this with the following program and both stderr and stdout make it out with the command "netconsole-host python3 testprint.py"

import sys
while True:
    print("hello world stdout")
    print("hello world stderr", file=sys.stderr)

However, due to buffering, the lines are NOT interleaved on a line-by-line basis.
If you want that, you have to append flush=True to the args to print() (or flush the stream yourself).

virtuald, 2015-01-11:
Aha! Very interesting. We should add this to the robotpy documentation somewhere then.

closed 2015-01-11 by virtuald

PeterJohnson, 2015-01-11:
Aha, adding "-u" to the python command line makes stdout unbuffered.  Not sure how this affects performance, though.

virtuald, 2015-01-11:
Cool, I had thought there was an option to do this somewhere. Presumably if netconsole-host is always reading from its pipe and not blocking on UDP, then we won't hit performance issues? Not sure how to measure that though.

virtuald, 2015-01-17:
I thought about this again, and looked into it. Apparently cpython will buffer writes into a pipe, but if it's attached to a TTY, then it will flush on each \n. If netconsole-host was to open a pty, it might work as expected.

virtuald, 2015-01-17:
More details: http://bugs.python.org/issue11633

Python3 enables line buffering, even when -u is passed. I think that would be good enough for most performance, right? I'm going to add it to pyfrc to do it by default, with an option to disable it.

virtuald, 2015-01-17:
Oh, interesting: https://github.com/robotpy/wpilibtools/commit/a699dfad8ebf0131c58627b1a01fee46327ae2a1

And... they write netconsole-host on each deploy. Presumably to make upgrades easier. Sigh.

https://github.com/robotpy/allwpilib/commit/e614217d416fa7e92d66236fc11a69e2baab8626

virtuald, 2015-01-18:
FYI, I added the -u to a deploy, and it didn't work. I suspect the new version of netconsole will be needed, will have to try it later.

virtuald, 2016-01-04:
FYI, for the 2016 image with updated netconsole-host, I ran into some stderr issues, but `-u` seems to fix them. Will change that and see how it goes.