robotpy/pyfrc #29: Timer breaks when using PIDController

sarosenb, 2015-02-08:
The PIDController uses a TimerTask object which also calls Timer.delay which increments our fake_time timer while running. Since it is running on its own thread, how many calls it makes is indeterminate and  if what you wanted to get done in that time is also indeterminate. I think this means that we shouldn't be doing tests on actual time or do something different if the object type is a timertask. I am not sure which action to do so wanted to have a short discussion.

sarosenb, 2015-02-08:
Other option is that fake timer increments on a sped up real timer based on the computers timer. that would avoid any fake time issues and allow a real delay for the threads.

virtuald, 2015-02-13:
I neglected to look at this, sorry. Did you end up finding a solution?

sarosenb, 2015-02-13:
I gave my opinions on what to do not sure what is a better option. It maybe as simple as not increment fake_time if the object type is a timer_task, but am hesitant to do that, wanted your opinion.

commit 74fa83ab (2016-01-07): Thread control added to tests.

Fixes #29
Fixes robotpy/robotpy-wpilib#116

Because of the faked calls to Timer.delay any child threads
(used by PID loops, for example) would run their target functions
repeatedly with no delays.
This patch intercepts the delay call and uses a blocking call to wait
for a threading.Event. This makes the threads sleep until woken by
the main thread incrementing the time.
When a child thread requests a positive timestep that means it should
fire in the future, so it is made to sleep and woken if the time
advancement goes past this time. Threads that request negati [... cut]
closed 2016-01-07 by virtuald