robotpy/pyfrc #36: Refactor physics component into robotpy-wpilib-utilities?

computer-whisperer, 2015-11-17:
The physics subsystem here could be improved, and I think it would be beneficial to allow access to `physics.py` from within user robot code. However, `physics.py` relies upon the physics functions bundled with pyfrc, and won't run on a robot. I would like to split those methods out of pyfrc and build an improved physics engine in robotpy-wpilib-utilities.

Thoughts?

virtuald, 2015-11-17:
Sure, have at it. I was hoping to reimplement the physics stuff for the websim in javascript anyways, so feel free to do what you think makes sense.

computer-whisperer, 2015-11-25:
I have most of the engine built, and just need to get it properly integrated with pyfrc. Do you think keeping backwards compatibility is particularly important? If so, I could work something into pyfrc to detect the difference between "physics.py" and something like "dynamics.py" (The newer version)

virtuald, 2015-11-30:
Sorry, was on vacation.

I think if it's not that much effort, I'd like to mostly keep backwards compat, unless there's a compelling reason not to.

closed 2016-01-18 by virtuald