robotpy/pyfrc #39: Test harness breaks when freeing CANTalon

james-ward, 2016-01-07:
If a CANTalon is a static member of a robot class (ie like the way subsystems are declared in command based robots), tests will fail in the teardown due to an index error.
For example:
```
import wpilib

class TrivialRobot(wpilib.IterativeRobot):
    motor = wpilib.CANTalon(1)

if __name__ == "__main__":
    wpilib.run(TrivialRobot)
```
Running `python3 robot.py test --builtin` results in:
```
handle = <hal_impl.types.TalonSRX object at 0x7fe70d533240>

    def c_TalonSRX_Destroy(handle):
>       del hal_data['CAN'][handle.id]
E       KeyError: 1
```

virtuald, 2016-01-07:
It is generally discouraged to create motor objects before robotInit is called.

james-ward, 2016-01-07:
Doesn't this mean that the command-based examples are wrong? They initialise their subsystems before robot_init is called.
Should we be changing the examples, in that case?

james-ward, 2016-01-07:
Gearsbot and pacbot, don't do this, but the example command-based robot does. I'll change it.

closed 2016-01-07 by james-ward

virtuald, 2016-01-07:
Yes, the examples should be changed then.

As I'm sure you've noticed, the problem is when you run multiple tests -- if the objects are initialized statically, they never get a chance to 'reset' for the new test because we can't reload the module (well.. maybe we can...). Unless we reinitialized all existing objects... but that sounds like a problem waiting to happen.