robotpy/pyfrc #40: Simulator's controllers inconsistent with driver station

Arhowk, 2016-01-10:
With a standard xbox controller, the axes are as follows

0) Left Rotate
1) Left Move
2) Left Trigger
3) Right Trigger
4) Right Rotate
5) Right Move

However, on the simulator with pygame, the axes are as follows

0) Left Rotate
1) Left Move
2) Both Triggers
3) Both Triggers
4) nothing
5) crashes

virtuald, 2016-01-10:
I don't own an Xbox controller, so I can't test it. Can you patch the pygame implementation so it's correct? The implementation seems to be pretty straightforward: https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/sim/pygame_joysticks.py

Arhowk, 2016-01-10:
Okay, I'll see if I can. However, in the meantime, sorry for being totally
off topic but how do you change the install/download target? I first
downloaded it to 172.22.11.2 but now I can't change it and I can't find
anything in the docs that mentions changing it.

Arhowk, 2016-01-10:
Okay there seems to be an issue in the wpilib for getting joystick values too. The values retrieved from the roboRIO were wildly incorrect as well, with 0 and 1 being the only two axes to match their driver station values.

Arhowk, 2016-01-11:
Where would I find this locally? there is no C:\Python34\Libs\pyfrc folder

virtuald, 2016-01-11:
It would be `C:\Python34\Libs\site-packages\pyfrc`. But even better, what I would do is the following:

    py -3 -m pip uninstall pyfrc
    git clone https://github.com/robotpy/pyfrc.git
    cd pyfrc
    py -3 -m pip install -e .

This will install pyfrc in 'develop' mode, which means any changes you make in the git repository will automatically show up when you do a 'import pyfrc'.

See [setuptools develop mode](https://pythonhosted.org/setuptools/setuptools.html#develop-deploy-the-project-source-in-development-mode) docs for more details.

virtuald, 2016-01-11:
Also, have you tried using [XBoxController](https://github.com/robotpy/robotpy-wpilib-utilities/blob/master/robotpy_ext/control/xbox_controller.py) object?

virtuald, 2016-01-11:
Good question. You can delete the .deploy_cfg file. Should probably have an option to change it.

Arhowk, 2016-01-11:
Yeah I found that file after some digging around. Maybe something in the
eclipse plugin, either a config option for "team IP address" or just a
"Deploy to new IP" would be good.

Also, after some tests, getRawAxis fixes my bug with the real robot but it
still shows the same issues in the simulator. XboxControiller did not solve
the issue either (its just a wrapper for getRawAxis anyway).

I looked in that directory and did not find pyfrc. Turns out that pyfrc
installed itself in the python 3.5 directory instead of the 3.4 (I
installed 3.4 over 3.5 because of issues with pygame), so I reinstalled
pyfrc in 3.4 and it seems to be correct now. I'll report back if I'm able
to find out why the discrepancy exists.

Also, for future reports, would you rather me communicate over the relevant
github repository or ChiefDelphi? I've been using github for a while so it
is pretty much second nature.

virtuald, 2016-01-11:
github is preferred. I have a note somewhere to make a team IP address option available.

Are you using the eclipse plugins? I had found they weren't very useful so I haven't upgraded them for 2016.

Arhowk, 2016-01-11:
Yes. I can't figure out how to make the "run" button actually work instead of using "run as" but other than that they are useful for me. I like to close out of tabs constantly so I end up closing the command line tab to deploy code and whatnot which is why I switched to the GUI version. (I also already had eclipse installed for work).

Arhowk, 2016-01-11:
So there was really no way to ACTUALLY fix the issue, because pygame uses DirectInput and the driver station uses XInput (that may be backwards... but you get the issue).

My fork of the repo has a workaround fix for this- https://github.com/Arhowk/pyfrc

Basically, it added 2 to the maximum axes count than checked if it hit the axes maximum, than its probably an xbox controller and it will move the axes around accordingly.

There are a few issues..

* Triggers dont behave EXACTLY like in the robot. In the robot, both triggers are independent. In the sim, you can only have one held at a time. I have no way to fix this but its quite a rare scenario in which an issue arises.
* Controllers with 5 axes automatically get setup to be xbox controllers. Most 6 axis controllers are xbox or xbox-variant controllers anyway so, again, not a huge issue.
* The UI doesn't show axes besides X and Y properly. Not a huge issue using the controller but it removes the opportunity for xbox controllers to be played with via the interface if you don't have an xbox controller

Arhowk, 2016-01-18:
Addition: current pysim has no POV support. (need todo)

virtuald, 2016-01-20:
Added POV support in 2016.1.3, but not in pygame support. The math is a bit messed up if you click it with a mouse, but I'm tired and don't use it. PR accepted. :)

Arhowk, 2016-01-20:
Hm? wpilib.Joystick already had a getPov() but it returned nothing on the
sim

virtuald, 2016-01-20:
If you upgrade, you'll see a POV thing you can click on in the simulator.

virtuald, 2017-02-25:
Closing due to age.

closed 2017-02-25 by virtuald