robotpy/pyfrc #43: Detect that the user doesn't have any tests, and default to running the builtin

virtuald, 2016-01-15:
Or, perhaps tell them that they should run the builtin.

virtuald, 2016-01-15:
Inspired by http://www.chiefdelphi.com/forums/showthread.php?p=1524487

ArthurAllshire, 2016-01-16:
Also, it would be nice if the library could detect that you had already run the tests on a particular code base using a checksum, and then not run them again when you are deploying, or trying to run them again.

virtuald, 2016-01-16:
@ArthurAllshire cool idea. Looks like there's a pytest plugin that might help with this: https://pytest.org/latest/cache.html

virtuald, 2016-01-18:
I decided to not automatically run the builtin tests, as the user might have accidentally added the tests to the wrong place. Instead, I've opted to interactively ask them if they want the tests to be created, and automatically create the tests directory + pyfrc_test.py if they say yes.

commit a7ba97a7 (2016-01-18): Testing improvements (fixes #43)

- Detect if no tests were collected and warn the user appropriately
- If there are no tests, ask the user if they would like to add builtin tests
closed 2016-01-18 by virtuald