robotpy/pyfrc #46: Simulated time goes backwards

virtuald, 2016-01-23:
An exception occurs at https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/_impl/timertask.py#L32 because the delay period is negative.

Is evident in https://github.com/robotpy/robotpy-wpilib-utilities/issues/10 -- but I've found that the example works on Linux without a problem, but fails (sometimes) on OSX 10.11. I presume there's a weird race condition that was introduced with https://github.com/robotpy/pyfrc/pull/38? 

@james-ward could you look at this maybe? I won't be able to do any robot stuff until Sunday/Monday.

james-ward, 2016-01-23:
I'm happy to look at it. What exception is it throwing? I can't reproduce the error at my end I'm afraid.

virtuald, 2016-01-23:
```
Exception in thread PIDTask0:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/wpilib/_impl/timertask.py", line 32, in run
    Timer.delay(wait_til - now)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/wpilib/timer.py", line 74, in delay
    hal.delaySeconds(float(seconds))
  File "<string>", line 3, in delaySeconds
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/hal_impl/functions.py", line 1794, in delaySeconds
    hooks.delaySeconds(s)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pyfrc/test_support/pyfrc_fake_hooks.py", line 24, in delaySeconds
    self.fake_time.increment_time_by(s)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pyfrc/sim/sim_time.py", line 92, in increment_time_by
    time.sleep(secs)
ValueError: sleep length must be non-negative
```

I contemplated it, and I think I know whats happening.

* Time is now T
* thread gets time value, adds N to it
* main thread sleeps; time is now `T + N + 1`
* thread gets time value which is `T + N + 1`
* thread tries to sleep `(T - N) - (T + N + 1)` -- since that is negative the exception gets thrown

james-ward, 2016-01-24:
I will have a look at this tomorrow (was moving house this weekend). I think it is the same issue that I fixed in the test harness - if I can implement the same thing in the sim harness it will stop child threads from trying to increment the fake time and causing this problem.
I'll have to look deeper to be sure - I haven't really explored how the simulator works. Given that the documentation at the top of `FakeRealTime` says "Currently, we assume all robot code runs in a single thread. This makes a lot of things easier. If that assumption was broken, then this would be a bit more complex" I think that is probably the issue!

virtuald, 2016-01-24:
Haha, I had totally forgotten about that.

virtuald, 2016-01-26:
Hm, I looked a little bit at this last night, and created this [robot.py](https://gist.github.com/virtuald/417f03b63edbe0449361) program, and wasn't able to reproduce the problem on Linux.

virtuald, 2016-01-29:
I'm not able to reproduce anymore, and I've just modified the TimerTask to notice if the delay is negative and issue a warning. Will have to come back to this at a later date.

james-ward, 2016-02-15:
I've been digging around with this, because I think your quick fix is now stopping TimerTasks from being terminated nicely in the test harness.
My previous patch for the test (not sim) time stuff meant that when a thread made a call to `time.delay()` that was negative, it would be intercepted in the `FakeTime` class and would wait for the main thread to advance time until its delays were positive again. This meant that threads under test didn't run at a million miles an hour when in the test harness and fail to terminate properly.
The issue here is in the sim harness, when it is running in real time. The call on https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/sim/sim_time.py#L92 is failing because this is a call from a thread for negative time, which is being sent to the system `time.delay()` and not faked by the test harness.
I think the fix will be to check for negative time in `sim_time.py` - otherwise we get errors with the unit test harness hanging. My team is having problems with this when unit testing a robot with PID loops.

I'm afraid this solution might need to be verified by someone else - I don't have access to OSX, and that seems to be the OS affected by the bug.

virtuald, 2016-02-15:
Are you able to reliably reproduce the hanging? I just tried locally (linux) the pacgoat and gearsbot examples, and neither of them hung. I've got time this week to dig into this if it can be reproduced.

As an interesting experiment, I tried calling `wpilib.Timer.delay(-7)` on a real roborio, and it [completely hangs (can't even CTRL-C)](https://github.com/robotpy/robotpy-wpilib/issues/181). With that in mind, I think it's important to ensure that calls to Timer.delay() can never be negative.

virtuald, 2016-02-23:
Well, Travis-CI is able to reproduce the hanging: https://travis-ci.org/robotpy/robotpy-wpilib-utilities/jobs/111145166

I haven't been able to reproduce it locally.

virtuald, 2020-03-24:
OBE

closed 2020-03-24 by virtuald