robotpy/pyfrc #56: Deploy issue when network uses IPv6

ryannazaretian, 2016-05-03:
File: pyfrc/lib/pyfrc/robotpy/installer.py

Code (line 366 through 379):

```python
if not no_resolve:
    try:
        print("Looking up hostname", hostname, '...')
        addrs = socket.getaddrinfo(hostname, None)
    except socket.gaierror as e:
        raise Error("Could not find robot at %s" % hostname) from e
        
    # pick the first address that is sock_stream
    for _, t, _, _, (ip, _) in addrs: # <- Error on this line due to (ip, _) not working for IPv6
        if t == socket.SOCK_STREAM:
            print("-> Found %s at %s" % (hostname, ip))
            print()
            hostname = ip
            break 
```

The returned tuple from socket.getaddrinfo() is different depending on if it is IPv4 or IPv6 addresses. 

socket.getaddrinfo() will return two differently formatted tuples, depending on if the address is IPv4 or IPv6:

`IPv4: [(<AddressFamily.AF_INET: 2>, 0, 0, '', ('8.8.8.8', 0))]`
`IPv6: [(<AddressFamily.AF_INET6: 23>, 0, 0, '', ('2001:4860:4860::8888', 0, 0, 0))]`

I do not currently have a system to test with, so this might not be the best way. I believe everything in the control system supports IPv6 since our RoboRIO was resolving with a IPv6 address. I'm not sure if the ssh supports IPv6 or what other implications there are, but the RoboRIO still had an IPv4 address associated with it:

socket.getaddrinfo('roborio-4901-frc.local', None) returns
`[ (<AddressFamily.AF_INET6: 23>, 0, 0, '', ('fe80::280:2fff:fe17:3af1%8', 0, 0, 8)), (<AddressFamily.AF_INET: 2> 0, 0, '', ('129.252.23.54', 0)) ]`

Per the Python socket.getaddrinfo() documentation:

> The function returns a list of 5-tuples with the following structure:
> (family, socktype, proto, canonname, sockaddr) 

How about, instead of unpacking it like this:
```python
for _, t, _, _, (ip, _) in addrs:
    if t == socket.SOCK_STREAM:
        print("-> Found %s at %s" % (hostname, ip))
        print()
        hostname = ip
        break 
```

You unpack it like this:
```python
for _, t, _, _, sockaddr in addrs:
    if t == socket.SOCK_STREAM:
        print("-> Found %s at %s" % (hostname, sockaddr[0]))
        print()
        hostname = sockaddr[0]
        break 
```

If IPv4 doesn't work, you can check the first tuple element to see if it is an AF_INET or AF_INET6:
```python
for family, t, _, _, sockaddr in addrs:
    if family == socket.AddressFamily.AF_INET and t == socket.SOCK_STREAM:
        print("-> Found %s at %s" % (hostname, sockaddr[0]))
        print()
        hostname = sockaddr[0]
        break 
```

virtuald, 2016-05-03:
Or try to collect both, and prefer ipv4 if it's found? But, you're right about fixing the unpacking. If you submit a PR I'll test/merge it, otherwise I'll try to get to this in the near future.

One of the other problems with using getaddrinfo() is that if you are trying to use an ssh alias in `~/.ssh/config` then of course getaddrinfo doesn't work.

ryannazaretian, 2016-05-04:
I think I'm going to sort the addresses using the AddressFamily and pick the first index, which will be the IPv4 address, if it exists.

closed 2016-08-13 by virtuald