robotpy/pyfrc #70: The default tests do not call many methods that they should

Bobobalink, 2017-02-06:
Running `python robot.py coverage test` does not cover many important methods, such as `robot.AutonomousInit()`, `robot.TeleopInit()`, and the `execute()` method of commands that are the defaults  of subsystems, although their `__init__()` methods are called.

virtuald, 2017-02-06:
It will only cover the things that your tests cover. The default tests aren't meant to be comprehensive, if you need more coverage then you should create custom tests.

virtuald, 2017-02-06:
Also, `autonomousInit` and `teleopInit` do not start with a capital letter.

virtuald, 2017-02-06:
See https://github.com/robotpy/examples/blob/master/sample/tests/basic_test.py for an example of how to setup custom tests.

I'm open to suggestions and PR's for increasing the default test coverage, but it seems like a rather hard problem -- every robot is a bit different, how would the tests know what to cover?

Bobobalink, 2017-02-06:
I don't really understand how the testing system works, but shouldn't
```
    control.set_autonomous(enabled=True)
    control.run_test(lambda tm: tm < 15)
```
(in basic_tests.py) at least call `autonomousInit()` and run the robot as if it were in autonomous, even if it does not recieve any inputs?
If not, how do I go about doing that?

virtuald, 2017-02-06:
It should call `autonomousInit`. Most of the logic lives in controller.py... you can see that when run_test is called, then `startCompetition` is called: https://github.com/robotpy/pyfrc/blob/master/lib/pyfrc/test_support/controller.py#L148

For a command based robot, `startAutonomous` is here: https://github.com/robotpy/robotpy-wpilib-utilities/blob/master/commandbased/commandbasedrobot.py#L14, and as you can see it calls autonomousInit et al for you. If it doesn't call them... that could be an actual bug.

Bobobalink, 2017-02-06:
Testing does not appear to run the methods that it should be. Running the simulator does call the methods, and appropriately logs that it ran `autonomousInit()` and `telelopInit()`, and the coverage report agrees.

virtuald, 2017-02-06:
Looks like `SampleRobot` was getting tested properly, but `IterativeRobot` isn't. I suspect it broke because of the 2017 HAL changes related to the DriverStation mutex. I've got a set of unit tests that will detect such failures in the future.

Upgrade pyfrc and your issue should be resolved.

commit 5c95f48a (2017-02-06): Testing: signal the DS that a 'new packet' has been received

- Required for IterativeRobot
- Fixes #70
closed 2017-02-06 by virtuald