robotpy/pyfrc #71: Simulator exception if there's multiple PCMs

vScourge, 2017-02-21:
Using pyfrc 2017.0.7
If the robot code utilizes multiple Pneumatics Control Modules, the simulator throws an exception when Solenoid objects are created, thinking the same ID resource has been used more than once.  For example, if we have one PCM assigned to CAN ID 0 and a second one at CAN ID 2, this line will work:

`solenoid1 = wpilib.Solenoid( moduleNumber = 0, channel = 0 )`

But this will then throw an error:

`solenoid2 = wpilib.Solenoid( moduleNumber = 2, channel = 0 )`

The exception:
```
File "C:\Python34\Lib\site-packages\wpilib\solenoid.py", line 75, in __init__
  self._solenoidHandle = hal.initializeSolenoidPort(portHandle)
File "C:\Python34\Lib\site-packages\hal\functions.py", line 42, in outer
  raise HALError(getErrorMessage(status.value))

hal.exceptions.HALError: HAL: A resource is already allocated.
```

Deploying the same code to the robot works fine.  Seems to be unique to the simulator.

virtuald, 2017-02-22:
Yeah, the simulated HAL definitely only anticipated people using a single PCM. I'm not 100% sure the best way to accommodate a second PCM.

I guess what could happen is we could create a PCM key, and then make the solenoid key be equal to PCM[0]... that way we wouldn't break backwards compatibility.

vScourge, 2017-02-22:
So the key is a combo of module number + channel?  But module number defaults to zero.  Sounds reasonable to me.

virtuald, 2017-02-22:
What I imagine one would end up with is this:

* `hal_data['pcm'][X][Y]` - referring to individual PCM/solenoid combinations
* `hal_data['solenoid'][X]` - is actually PCM[0]

This would allow advanced users to use as many PCMs as they require, while keeping solenoid access simple for those with a single default PCM.

closed 2017-02-25 by virtuald