robotpy/pyfrc #76: Add pyflakes tests

virtuald, 2017-03-10:
pyflakes only detects things that are actually code errors (well... mostly), and it's pretty fast. Should add this in the future.

I think I would have it separate from the tests, as most of the things that pyflakes catches are things that are almost certainly errors.

auscompgeek, 2023-01-02:
Note that pyflakes will emit warnings such as these too:

```
tests/pyfrc_test.py:6:1: 'from pyfrc.tests import *' used; unable to detect undefined names
tests/pyfrc_test.py:6:1: 'pyfrc.tests.*' imported but unused
```

`flake8` assigns those error codes F403 and F401 respectively. `ruff` emits the first error, but not the second for some reason (which is probably a bug).

auscompgeek, 2025-02-21:
Should we recommend teams install https://github.com/businho/pytest-ruff instead?

closed 2026-04-20 by virtuald