robotpy/pyfrc #81: Iteration issue with the test harness

virtuald, 2017-05-10:
Reported by @Twinters007 

```
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/pyfrc/tests/basic.py:107: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/pyfrc/test_support/controller.py:148: in run_test
    self._robot.startCompetition()
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/wpilib/samplerobot.py:142: in startCompetition
    self.disabled()
robot.py:49: in disabled
    wpilib.Timer.delay(0.010)
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/wpilib/timer.py:74: in delay
    hal.sleep(seconds)
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/hal_impl/functions.py:125: in sleep
    hooks.delaySeconds(s)
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/pyfrc/test_support/pyfrc_fake_hooks.py:24: in delaySeconds
    self.fake_time.increment_time_by(s)
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/pyfrc/test_support/fake_time.py:215: in increment_time_by
    self._wake_children(time)
/home/travis/virtualenv/python3.5.2/lib/python3.5/site-packages/pyfrc/test_support/fake_time.py:93: in _wake_children
    child_threads = list(self._child_threads.items())
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
self = <WeakKeyDictionary at 0x7fbe41e7c438>
    def items(self):
        with _IterationGuard(self):
>           for wr, value in self.data.items():
E           RuntimeError: dictionary changed size during iteration
```

I think the only viable way to prevent the dictionary from changing is by adding a separate lock, calling `weakref.finalize` on any thread we add to the dictionary, and then take that lock during finalization? Hm, but then we would need to just make this a normal dictionary. I wonder if there's a way to take a lock on the underlying dictionary in WeakKeyDictionary?

virtuald, 2017-05-10:
Might be able to use `WeakKeyDictionary.keyrefs()` instead of items.

virtuald, 2018-02-09:
Haven't seen it this year, will assume that #91 fixed it.

closed 2018-02-09 by virtuald

ariovistus, 2018-02-21:
cough 
https://travis-ci.org/Retro3223/robot2018/builds/344164912?utm_source=github_status&utm_medium=notification

virtuald, 2018-02-21:
That's a different issue. That indicates something is changing the dictionary of liveWindow components without taking the mutex.