robotpy/pyfrc #82: Invert Y axis of joystick

Twinters007, 2017-07-15:
I'm wondering if there's an easy way to invert the 'Y' axes values of the joysticks in the simulator automatically, because a value of 1 in the simulator intuitively should make the robot move 'forward', but because most joysticks have negative 1 at the top, teams negate this value in code, so the robot moves backward. 

I've created a hot fix by adding jaxes[1]*=-1  in ui.py

virtuald, 2018-02-08:
Why shouldn't it reflect the reality? Otherwise, a user might get confused as to why it's backwards in the simulator.

twinters007, 2018-02-08:
A value of `1` on a real joystick is pulling the joystick. In the simulator, it acts the same way, but if I see +1 for a joystick,  the robot should move forward in my mind. If we negate the value, then +1 in the sim relates to pushing the joystick forward, rather than the true value of positive 1

virtuald, 2018-02-08:
Yes, but your natural intuition is incorrect, because the joystick doesn't work that way. The simulator reinforces that and reminds you that it's backwards.

twinters007, 2018-02-08:
But having it this way in the simulator may cause teams to not invert the y axis of the joystick. So when they get to their robot, it drives backward

virtuald, 2018-02-09:
Then what we need is a better way of communicating that it will be backwards (maybe an image of a joystick)? Changing the value to be backwards will be confusing if they ever print out the actual value.

amboyscout, 2018-02-28:
Allowing axis inversion for visual purposes in the sim would be very nice. I don't think it should affect the HAL or anything of the sort, but I support Tim that we should be able to invert what is being displayed.

RustyRaptor, 2019-01-20:
Yes this would definitely help. We found out today the hard way that joysticks have their Y inverted like a flight stick. So yeah even something in the docs would be good.

virtuald, 2019-01-20:
I suppose a documentation note is good. But as I said, if you plug in a real robot and a real joystick, pushing 'forward' on a joystick will cause a -1 value. I don't see why the simulator shouldn't do the same thing.

virtuald, 2019-01-20:
What if we changed the visual of the joystick to be a square that you clicked? And it would vary x/y according to where you're clicking? -1 would still be forward, of course.

ArchdukeTim, 2019-01-20:
@virtuald I'd propose a circle rather than a square

auscompgeek, 2019-01-21:
@ArchdukeTim Actual joysticks are squares, not circles. Gamepads are weird.

auscompgeek, 2019-01-21:
It'd probably be good to have that as an option too, seeing as teams do use gamepads.

RustyRaptor, 2019-01-22:
Too bad you can't square a circle.

virtuald, 2020-03-24:
OBE

closed 2020-03-24 by virtuald