robotpy/pyfrc #84: robot simulator on high dpi devices

ariovistus, 2017-09-17:
running the pygame based simulator on a high dpi laptop, the window is really small. Is there a way to increase the size?

I tried something like

```
    root.tk.call('tk', 'scaling', 4.0)
```

inside pyfrc/sim/ui.py, but it seemed to have no effect

virtuald, 2017-09-17:
Honestly, I don't know enough about tk to know the answer to this -- though that solution showed up in my very brief google search for an answer. Let us know if you figure it out!

auscompgeek, 2017-10-14:
From what I can tell, that solution works in the exact opposite direction to what I expected. I have no idea how Tk works either, so who knows what's going on there.

auscompgeek, 2017-10-16:
It looks like IDLE do things on HiDPI displays now: https://bugs.python.org/issue31500

vanjac, 2018-03-15:
We have the opposite problem. A few people have laptops with smaller/lower resolution screens (like 1366x768) and the simulator doesn't fit on the screen. You can't see the game data at all, for example.

ariovistus, 2018-03-15:
I did end up finding a workaround for high dpi on fedora:
 install xpra, xorg-x11-server-Xvfb, then this script: https://github.com/kaueraal/run_scaled will scale the sim ui to a usable size

virtuald, 2020-03-24:
OBE

closed 2020-03-24 by virtuald