robotpy/pyfrc #87: Lag during initialization of teleop and autonomous

adam-mcdaniel, 2017-11-05:
Our team, 6517, played in the battle for the bluegrass preseason competition yesterday. We had switched to python 2 weeks prior, and it was working out pretty well. When each match began though, our autonomous wouldnt start for about 5 seconds after the autonomous period began, and the same for teleop. Is there a problem with robotpy and pyfrc or do i need to comb through my code?

virtuald, 2017-11-05:
That would be a problem with your code, many other teams have successfully used RobotPy without any reported issues last year.

My psychic debugging powers tells me that you created a Gyro object at the beginning of teleop and/or autonomous mode (gyro initialization has a 5 second sleep in it). That would cause a similar issue if you did the same thing in Java/C++.

virtuald, 2017-11-05:
Also, if you're having trouble identifying the issue, feel free to post a link to it (or a gist) and I'd be happy to help you debug it.

adam-mcdaniel, 2017-11-07:
Bingo, that was our problem. Thanks! Your help is very much appreciated :)

closed 2017-11-07 by virtuald