robotpy/pyfrc #107: simulator not simulating 20 mSec DS packet arrival times

dangyogi, 2018-01-29:
What it does:

When I run an IterativeRobot in the simulator, it runs my periodic functions as fast as possible.

What I expect:

It would not run the periodic functions faster than every 20 mSec.

vanjac, 2018-01-31:
I'm having this problem also.... it was working fine in an earlier version of the 2018 simulator.

virtuald, 2018-01-31:
I will have a lot of free time in February, so I can look into it then.

virtuald, 2018-02-08:
https://github.com/robotpy/robotpy-wpilib/pull/437 and https://github.com/robotpy/pyfrc/pull/111 fix this issue.

Update pyfrc and robotpy-wpilib and you should no longer see this issue.

closed 2018-02-08 by virtuald

vanjac, 2018-02-08:
Thank you, it works now. It also seems to start up much faster.

virtuald, 2018-02-08:
Blame @auscompgeek for the fixes. 👍

auscompgeek, 2018-02-08:
Heh, I didn't realise I accidentally fixed this. I forgot this issue was even filed.