robotpy/pyfrc #109: Motors won't start until joystick pushed to 1 or -1 value.

RustyRaptor, 2018-01-31:
I have tried manually entering floats less than 1 and it works so it can't be an issue with the motor controllers. Driver station shows proportional values. It seems to me like the joystick call is returning an integer instead of a float and it's converting anything less than 1 to 0. Not sure yet if it's a driverstation issue. I am using QDriverStation. Will try later with the official driver station on windows.

twinters007, 2018-01-31:
Are you talking about the simulator?

RustyRaptor, 2018-01-31:
No, the actual robot. Also I found this. Could this be where it's turned into an integer?

```
def getStickAxis(self, stick: int, axis: int) -> float:
```

twinters007, 2018-01-31:
Probably not. That returns the value of a given axis on a given stick.

virtuald, 2018-01-31:
That's just an annotation. Do you have code that reproduces the issue?

RustyRaptor, 2018-01-31:
Here is the code I am using on the robot

```
#!/usr/bin/env python3

import wpilib
from wpilib.drive import DifferentialDrive

class MyRobot(wpilib.SampleRobot):

    def robotInit(self):
        self.frontLeft = wpilib.Spark(0)
        self.rearLeft = wpilib.Spark(1)
        self.left = wpilib.SpeedControllerGroup(self.frontLeft, self.rearLeft)

        self.frontRight = wpilib.Spark(2)
        self.rearRight = wpilib.Spark(3)
        self.right = wpilib.SpeedControllerGroup(self.frontRight, self.rearRight)

        self.drive = DifferentialDrive(self.left, self.right)
        self.stick2 = wpilib.Joystick(0)
        self.stick = wpilib.Joystick(1)

    def disabled(self):
        while self.isDisabled():
            wpilib.Timer.delay(0.01)

    def operatorControl(self):
        timer = wpilib.Timer()
        timer.start()
        i = -1
        while self.isOperatorControl() and self.isEnabled():
            # if self.stick.getRawButton(2):
            #     # self.drive.arcadeDrive(self.stick.getY(), self.stick.getX())
            #     self.drive.arcadeDrive(1, 0)
            #     # self.drive.tankDrive(self.stick.getY() * -0.7, self.stick2.getY() * -0.7, True)
            # else:
            #     # self.drive.arcadeDrive(self.stick.getY(), self.stick.getX())
            #     self.drive.arcadeDrive(1, 0)
            #     # self.drive.tankDrive(self.stick.getY() * 0.7, self.stick2.getY() * 0.7, True)
            # # i = i * self.stick.getRawAxis(4)
            # # Move a motor with a Joystick
            # self.drive.arcadeDrive(self.stick2.getY(), 0, False)
            self.drive.arcadeDrive(self.stick2.getY(), 0, False)

            wpilib.Timer.delay(0.02)

    def autonomous(self):
        timer = wpilib.Timer()
        timer.start()
        while self.isAutonomous() and self.isEnabled():
            if timer.get() < 3.0:
                self.drive.tankDrive(-0.1, -1)
            else:
                self.drive.tankDrive(0., 0)

            wpilib.Timer.delay(0.01)

if __name__ == '__main__':
    wpilib.run(MyRobot)
```

Here is another one I tried 

```
#!/usr/bin/env python3
'''
    This is a demo program showing the use of the RobotDrive class,
    specifically it contains the code necessary to operate a robot with
    tank drive.
'''

import wpilib
from wpilib.drive import DifferentialDrive

class MyRobot(wpilib.IterativeRobot):

    def robotInit(self):
        '''Robot initialization function'''

        # object that handles basic drive operations
        self.frontLeftMotor = wpilib.Spark(0)
        self.rearLeftMotor = wpilib.Spark(1)
        self.frontRightMotor = wpilib.Spark(2)
        self.rearRightMotor = wpilib.Spark(3)

        self.left = wpilib.SpeedControllerGroup(self.frontLeftMotor, self.rearLeftMotor)
        self.right = wpilib.SpeedControllerGroup(self.frontRightMotor, self.rearRightMotor)

        self.myRobot = DifferentialDrive(self.left, self.right)
        self.myRobot.setExpiration(1.0)

        # joysticks 1 & 2 on the driver station
        self.leftStick = wpilib.Joystick(0)
        self.rightStick = wpilib.Joystick(1)

    def teleopInit(self):
        '''Executed at the start of teleop mode'''
        self.myRobot.setSafetyEnabled(True)

    def teleopPeriodic(self):
        '''Runs the motors with tank steering'''
        # self.myRobot.tankDrive(self.leftStick.getY() * -1, self.rightStick.getY() * -1)
        self.myRobot.tankDrive(0.2, 0.2, False)

if __name__ == '__main__':
    wpilib.run(MyRobot)
```

twinters007, 2018-01-31:
Since you're using QDriverStation, I would bet the issue is somewhere in there. Try it with a real driverstation. It may also help to open up the gamepad utility that comes with windows (search for 'set up usb gamepad') and make sure the values it's showing match what you expect.

RustyRaptor, 2018-01-31:
Yeah I also think that could be the case because when I do it in the simulator with pygame as the joystick handler it works as expected and I get the right values when I print them repeatedly.

virtuald, 2018-01-31:
Yeah... see if you can duplicate with a real DS.

A helpful thing you can do to isolate this issue is just create a very simple program that only sends the joystick values to SmartDashboard and does nothing else, and view the values there. If the values are wrong with a real DS, then it's probably a RobotPy bug. If they're correct, then it's something else.

```
def teleopPeriodic(self):
    SmartDashboard.putNumber("joy", self.stick.getY())
```

RustyRaptor, 2018-01-31:
I've also noticed that QDriverStation isn't being updated anymore. That's unfortunate. I hope someone adopts the project and updates it with the latest protocol. That's the only driver station that runs on Linux. NI isn't very nice about source code sadly.

RustyRaptor, 2018-01-31:
Ok I've confirmed it's QDriverStation. Will submit an issue there.

closed 2018-01-31 by virtuald