robotpy/pyfrc #115: AttributeError: 'StringVar' object has no attribute 'trace_add'

dangyogi, 2018-02-12:
Trying to start program with simulator.  pyfrc version 2018.1.1, wpilib version 2018.1.2.  This worked in pyfrc version 2018.0.5

  File "robot.py", line 50, in <module>
    wpilib.run(ModularRobot)
  File "/home/bruce/uCR/FRC/FRC_2018/venv-test/lib/python3.5/site-packages/wpilib/_impl/main.py", line 136, in run
    retval = options.cmdobj.run(options, robot_class, **kwargs)
  File "/home/bruce/uCR/FRC/FRC_2018/venv-test/lib/python3.5/site-packages/pyfrc-2018.1.1-py3.5.egg/pyfrc/mains/cli_sim.py", line 62, in run
    ui = sim.SimUI(sim_manager, fake_time, config_obj)
  File "/home/bruce/uCR/FRC/FRC_2018/venv-test/lib/python3.5/site-packages/pyfrc-2018.1.1-py3.5.egg/pyfrc/sim/ui.py", line 52, in __init__
    self._setup_widgets(frame)
  File "/home/bruce/uCR/FRC/FRC_2018/venv-test/lib/python3.5/site-packages/pyfrc-2018.1.1-py3.5.egg/pyfrc/sim/ui.py", line 356, in _setup_widgets
    self.gamedataval.trace_add('write', self.on_gamedata_selected)
AttributeError: 'StringVar' object has no attribute 'trace_add'

dangyogi, 2018-02-12:
Should have mentioned, this is python 3.5 with tkinter version 8.6.

auscompgeek, 2018-02-13:
Hm. Which OS/distro, and how was Python installed?

dangyogi, 2018-02-13:
Ubuntu 16.04, the python that comes installed, I think.

virtuald, 2018-02-13:
Sorry about that, 2018.1.2 should fix it.

commit 0231db1a (2018-02-13): trace_add is a new tk api, use deprecated trace_variable when not present

- Fixes #115
closed 2018-02-13 by virtuald