robotpy/pyfrc #117: Make it easier to control tank drive robots in simulation

virtuald, 2018-02-14:
Usually the way I work around this is adding an 'isSimulation' check which switches to arcade drive, but then it doesn't exercise the tank drive path. Would be nice to just provide a way to bind two joystick inputs together to make them easier to control.

twinters007, 2018-02-14:
The only problem with this is getting it to turn since you need them to be opposite. Maybe it'd be better to have the keyboard act as a joystick with wasd and ijkl?

virtuald, 2018-02-14:
I meant bind them together in such a way that it was easy to control. A keyboard based joystick would be cool too.

twinters007, 2018-02-14:
Instead of the arcadeDrive in the robot code, maybe a circular joystick widget that converts arcade to tank?

virtuald, 2018-02-14:
Exactly.

amboyscout, 2018-02-28:
Tank drive in the sim is definitely a pain. I'd say the keyboard solution wouldn't work too well unless you have laser switches. Circular joystick widgets would be cool in general and may help to remedy this problem. Another option for users would be to swap the joystick numbers during sim to be right next to each other for easier tank drive. Along with this, what is the possibility of fixing the joystick inputs to work with typed in numbers? As of now, when you click on a joystick and type a number, 0 sets 0 and 1 sets 1, but I cant get in between by typing numbers in. This would make certain tank drive testing, and testing in general, easier because fine-tuning of the values you input helps a lot.

virtuald, 2018-02-28:
I don't have much time to work on this at the moment, but I welcome potential fixes. The sim stuff all lives in https://github.com/robotpy/pyfrc/tree/master/lib/pyfrc/sim

auscompgeek, 2022-09-11:
OBE. We no longer have our own simulation GUI.

closed 2022-09-11 by auscompgeek