robotpy/pyfrc #150: Put simulator motor outputs in a scrollbox

vanjac, 2019-01-17:
With too many motors the simulator window can become too big to fit on the screen.

virtuald, 2019-01-17:
How many motors are we talking about?

auscompgeek, 2019-01-17:
My team had 11 Talon SRXs on the robot, and a bunch of us struggled to use the whole simulator window on some of our laptops.

vanjac, 2019-01-17:
Around 12 for us right now

vanjac, 2019-01-17:
Lots of team members have to use their school laptops with 13 inch screens

vanjac, 2019-01-17:
The joystick interface is also pretty tall, it might help to condense it

auscompgeek, 2019-01-23:
Another data point: one of my students is unable to use the reset button with 8 Talon SRXs on screen.

virtuald, 2019-02-05:
The latest release of robotpy-ctre wraps the motors into multiple columns, which should help some. We do need a better solution though.. perhaps the websim would be better?

vanjac, 2019-02-06:
The updated version is much better. Now it's actually too wide to fit on our school laptops. I've attached a screenshot.

Something I immediately notice is that the Solenoid display could be moved below the Digital section.

[image]

virtuald, 2020-02-17:
OBE

closed 2020-02-17 by virtuald