robotpy/pyfrc #151: Unicode error when running create physics.

Android5613, 2019-01-19:
I am running this on the newest version of pycharm community and the physics file generator will not work and gives me the error codes below. The operating system I use is windows 10.
I am trying to make a physics file so we can run our mecanum drive robot on the simulator.
```
C:\Users\Yueng\AppData\Local\Programs\Python\Python37\python.exe C:/Users/Yueng/PycharmProjects/5613-mecanum-2019A/robot.py create-physics
14:53:49:249 INFO    : wpilib              : WPILib version 2019.0.1
14:53:49:249 INFO    : wpilib              : HAL base version 2019.0.1; sim platform version 2019.0.1
14:53:49:249 INFO    : wpilib              : Running with simulated HAL.
Traceback (most recent call last):
  File "C:/Users/Yueng/PycharmProjects/5613-mecanum-2019A/robot.py", line 57, in <module>
    wpilib.run(MyRobot)
  File "C:\Users\Yueng\AppData\Local\Programs\Python\Python37\lib\site-packages\wpilib\_impl\main.py", line 198, in run
    retval = options.cmdobj.run(options, robot_class, **kwargs)
  File "C:\Users\Yueng\AppData\Local\Programs\Python\Python37\lib\site-packages\pyfrc\mains\cli_create_physics.py", line 109, in run
    fp.write(physics_starter)
  File "C:\Users\Yueng\AppData\Local\Programs\Python\Python37\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u03c9' in position 1436: character maps to <undefined>
```

auscompgeek, 2019-01-20:
Ugh, why are there non-ASCII characters in there.

ArchdukeTim, 2019-01-20:
Cause equations

auscompgeek, 2019-01-20:
I don't see any equations in the physics_starter.

RustyRaptor, 2019-01-20:
Made a PR for it. #152