robotpy/pyfrc #163: Robotpy documentary on Simulation Physics

Android5613, 2020-01-16:
When we were making a physics.py file we found out that how to access the motor values from the hal_data dictionary was not included, and that instead there was a todo saying to get the motor values from hal_data. My mentor figured it out so I wanted to fix it but I do not know how. 

The solution is the addition of these lines of code in the update_sim function:
lr_motor = hal_data["pwm"][1]["value"]
rr_motor = hal_data["pwm"][2]["value"]
lf_motor = hal_data["pwm"][3]["value"]
rf_motor = hal_data["pwm"][4]["value"]

auscompgeek, 2020-01-16:
My understanding is that the simulator here, including the robotpy-hal-sim package and its `hal_data` dict, is disappearing for this year.

virtuald, 2020-01-16:
Yes, we are currently exploring the right way to replace them this year.

closed 2020-01-21 by Android5613