robotpy/pyfrc #169: Demo instructions do not work

Pipeliner, 2020-02-22:
https://github.com/robotpy/pyfrc#quick-install--demo section says that after installing pyfrc with pip I should be able to run 
```
cd samples/physics/src/
python3 robot.py sim
```
However, there are no such directories nor files in the repo.

virtuald, 2020-02-22:
Due to significant changes in RobotPy this year, the documentation is a bit out of date. https://robotpy.readthedocs.io/en/stable/2020_notes.html#where-is-physics-and-tests now has a link to our single working physics demo. Sorry for the inconvenience.

Pipeliner, 2020-02-22:
Thank you for quick reply!
Actually I was figuring out what exactly this project is so I'm not really interested in *physics* demo, I'd be happy to see *any* demo.

Anyway, running the linked https://github.com/robotpy/examples/blob/master/physics/tests/pyfrc_test.py file didn't produce any output (after I completed site-packages/pyfrc/config.py
 so that it does not throw KeyError's).

virtuald, 2020-02-22:
Yeah, the test support still doesn't work, just the simulation physics. You can run `python robot.py sim` and the robot can be driven on the field, and there's some motor/digitalinput interactions modeled.

commit 84871c7d (2020-02-22): Remove out of date instructions

- Fixes #169
closed 2020-02-22 by virtuald