robotpy/pyfrc #171: deploy: deal with out of disk space conditions

virtuald, 2020-03-03:
Might want to provide a 'wipe' command to delete user code from the roborio, in case they accidentally fill up their disk?

Reference: https://www.chiefdelphi.com/t/cant-deploy-python-code/380350

benjiboy50fonz, 2021-01-21:
What are you looking for exactly? Something like a shell script? Or an actual command, and if so, where?

virtuald, 2021-01-21:
I suspect the best way to solve the problem mentioned in that thread is to make `./robot.py undeploy` exist, which would just open an SSH connection and execute `rm -rf /home/lvuser/py`. Probably can copy a lot of pieces from the deploy command implementation.

benjiboy50fonz, 2021-01-22:
On it. Hopefully will have it done by tonight.

benjiboy50fonz, 2021-05-21:
Time to close?

closed 2021-05-21 by auscompgeek