robotpy/pyfrc #172: Cannot locate physics.py

kajchang, 2020-10-20:
When I run the robot simulator, `pyfrc` looks for `physics.py` in the wrong place.

```
14:51:42:025 WARNING : pyfrc.physics       : Cannot enable physics support, /usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/Resources/physics.py not found
```

virtuald, 2020-10-21:
How are you executing robot.py?

kajchang, 2020-10-21:
I'm just running `python3 robot.py sim` with the physics module in the same folder.

I'm on Mac and installed Python through Homebrew.

virtuald, 2020-10-23:
Sorry, I missed your response! Busy busy...

Please try https://github.com/robotpy/pyfrc/pull/173 and see if that resolves your issue.

kajchang, 2020-10-23:
Works for me. Thanks!

commit cb6600a1 (2022-01-17): Save current working directory when loading extensions

- Ensures that physics.py can be loaded, among other things
- Fixes #172
closed 2021-01-25 by virtuald