robotpy/pyfrc #194: Provide mechanism to pass git hash / other things as part of deploy

virtuald, 2022-02-15:
I'm thinking maybe a hook of some kind? 

Or, we could auto-generate a JSON file with useful information to `deploy.json`.

```
{'git-version': '1234abde', 'path': 'C:\User\GitRepo', 'username': "Bob", 'hostname': 'awesome-laptop'}
```

And print it out during robot startup if `deploy.json` is present?

KenwoodFox, 2022-02-15:
I think i found where this belongs: https://github.com/robotpy/pyfrc/blob/2022/pyfrc/mains/cli_deploy.py

Should we make a custom hooks system or just implement this single hook?

virtuald, 2022-02-15:
I haven't really thought through the requirements at this point. I think initially 

If we wanted the user to be able to customize the json, we could define the following method in the robot:

```
class MyRobot(wpilib.TimedRobot):
    @staticmethod
    def deploy_hook(data):
        data["something"] = something

```

If we did that... we could add a `deploy_data` method to the robot instance if the user wanted to access it? Maybe that's a bit much though. Should think about how this would actually be used.

KenwoodFox, 2022-02-15:
I like the idea of doing something custom with a similar format to how physics.py works, where its inehert and says physics is not enabled if the file does not exist, but runs physics sim when it is present.

perhaps a similar system with a hooks.py? excecuting a few methods like `deploy_hook()` and `test_hook()`? Code to be run on the host doing the developing, seperate from robotinit and stuff that is meant to be for the robot

KenwoodFox, 2022-02-15:
Could we make a `build_info.json` purely in deploy, never saving it to disk but passing it over sftp? it would keep it from showing up on the dev's machine. Though if you're using git to manage your robot, you'ld prolly add the json to your gitignore and saving it locally may not be a hassle.

virtuald, 2022-02-15:
If we went the json route, I have a strong preference for it only existing on the robot. We already make a temporary directory to copy data over, see https://github.com/robotpy/pyfrc/blob/c94b21b22de6f654ab866878fe8e1cd9ebef1f81/pyfrc/mains/cli_deploy.py#L388

KenwoodFox, 2022-02-15:
> If we went the json route, I have a strong preference for it only existing on the robot. We already make a temporary directory to copy data over, see
> 
> https://github.com/robotpy/pyfrc/blob/c94b21b22de6f654ab866878fe8e1cd9ebef1f81/pyfrc/mains/cli_deploy.py#L388

Oh, this is perfect. I 100% agree, i can modify #196 to reflect this.

KenwoodFox, 2022-02-15:
Not everyone uses git so it may be nice if it just ommits a git-revision rather than not inluding the other vars at all, that way code managed by other source control engines or not managed at all can still utilize the other values like hostname and build date that may be more helpful

auscompgeek, 2022-02-15:
Heh, I have a hack to include the git branch and commit hash in a deploy in my team's robot code. https://github.com/thedropbears/pyrapidreact/blob/main/utilities/git.py

virtuald, 2022-02-16:
It would definitely be a requirement to not fail if it wasn't a git repo.

@auscompgeek you've been holding out on us!

KenwoodFox, 2022-02-16:
Super cool @auscompgeek! does this require transfering the `.git` to the robot though?

auscompgeek, 2022-02-16:
> does this require transfering the `.git` to the robot though?

Nope! There's a few assumptions made here:

- On our laptops, the code is always run from the git repo, and git is available in the environment it runs in.
- The deploy skips dotfiles, hence `.git` is not present on the robot.
- The deploy does not care about gitignore.

KenwoodFox, 2022-02-16:
Interesting, so if its possible to run stuff like this at the module level on the dev laptop to generate artifacts for the bot, them #195 is not needed :3
ill still work on #196

virtuald, 2022-03-04:
Fixed in #196

closed 2022-03-04 by virtuald