robotpy/pyfrc #204: [BUG]: slow deploy

virtuald, 2022-04-02:
### Problem description

@TheTripleV reports using this fixed their slow deploy issues:

```
sock = socket.socket(socket.AddressFamily.AF_INET, socket.SOCK_STREAM)

retry_on_signal (lambda: sock.connect((self.hostname, 22)))
self.client.connect(
  self.hostname,
  username=self.username,
  password=self.password,
  allow_agent=False,
  look _for_keys-False,
  sock=sock,
)
```

TBH, probably should add the `retry_on_signal` to [RobotFinder](https://github.com/robotpy/robotpy-installer/blob/344c87b5e9590ea5d3a2633a7ae66ca984ae5424/robotpy_installer/robotfinder.py#L39) and just reuse that socket instead of making a second connection?

### Operating System

Windows

TheTripleV, 2022-04-02:
Netconsole also has this issue.

virtuald, 2022-12-01:
I'm surprised this is an issue in newer pythons because of https://peps.python.org/pep-0475/

closed 2024-01-22 by virtuald