robotpy/pyfrc #220: Deploy fails because os.getlogin fails

TheTripleV, 2023-04-20:
On Linux mint 21.1, in both the is terminal app and the vscode terminal, deploy fails because the deploy info's os.getlogin() fails with no such device or address on Python 3.10.6.

auscompgeek, 2023-04-22:
From [the docs for os.getlogin](https://docs.python.org/3/library/os.html#os.getlogin):

> Return the name of the user logged in on the controlling terminal of the process. For most purposes, it is more useful to use [getpass.getuser()](https://docs.python.org/3/library/getpass.html#getpass.getuser) since the latter checks the environment variables LOGNAME or USERNAME to find out who the user is

So if the deploy is somehow run detached from a TTY it'll definitely error.

TheTripleV, 2023-04-23:
Yeah it was weird. We tried in the terminal app and in the vscode terminal and they both did that. I didn't look further. I patched his pyfrc to get rid of the call and hardcoded his name.

TheTripleV, 2023-09-28:
Closed by #221.

closed 2023-09-28 by TheTripleV