robotpy/pyfrc #229: [BUG] Cleanup doesn't occur correctly if tests fail

virtuald, 2024-01-28:
I think what's happening is that when pytest is processing the exception, the traceback holds references to local objects (including the robot) and so the robot and related objects (in some cases) are never fully freed, so when the hal handles and other things are cleared at the end of the test those objects are all orphaned. Several solutions come to mind:

* find a pytest hook that runs when we need it to happen and use that instead of at the end of the robot fixture
* Run the resets at the beginning of the next function instead
* We can technically run each tests in a new process now since the CLI doesn't have to import the robot code.. that would not be ideal, but many of these sorts of issues might disappear.

I believe this is causing https://github.com/robotpy/robotpy-navx/issues/30, but I don't have conclusive proof.

auscompgeek, 2024-01-28:
> I think what's happening is that when pytest is processing the exception, the traceback holds references to local objects (including the robot) and so the robot and related objects (in some cases) [...]

I concur here.

> We can technically run each tests in a new process now since the CLI doesn't have to import the robot code.

I've used pytest-forked previously, but that doesn't work on Windows.

Maybe it's time to solve #212 and use pytest-xdist?

auscompgeek, 2025-01-18:
An alternative: we could use the [`traceback`](https://docs.python.org/3/library/traceback.html) module to print stack traces without holding onto real exceptions. I guess we'd lose pytest's pretty handling of exceptions, but we probably don't need that for the robot tests?

I do like just solving the problem with xdist though.

virtuald, 2025-02-16:
Fixed in #238

closed 2025-02-16 by virtuald