robotpy/pyfrc #234: [BUG]: test_docstrings() returns internal error

PickleFace5, 2024-10-15:
### Problem description

When attempting to run the [test_docstrings](https://robotpy.readthedocs.io/projects/pyfrc/en/stable/testing.html#pyfrc.tests.docstring_test.test_docstrings) test using `robotpy test`, the test seems to return an internal error.

The test returns this:

```robot = <weakproxy at 0x000002683F6E2750 to Waffles at 0x000002683FA61C10>, robot_path = WindowsPath('C:/Users/caden/Documents/Steel Ridge Robotics/2024Waffles')

    def test_docstrings(robot, robot_path):
        """
        The purpose of this test is to ensure that all of your robot code
        has docstrings. Properly using docstrings will make your code
        more maintainable and look more professional.
        """

        # this allows abspath() to work correctly
        os.chdir(robot_path)

        errors = []

        for module in sys.modules.values():
>           if ignore_object(module, robot_path):

C:\Users\caden\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyfrc\tests\docstring_test.py:205:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _  

o = <module '_frozen_importlib' (frozen)>, robot_path = WindowsPath('C:/Users/caden/Documents/Steel Ridge Robotics/2024Waffles')

    def ignore_object(o, robot_path):
        """Returns true if the object can be ignored"""

        if inspect.isbuiltin(o):
            return True

        try:
            src = inspect.getsourcefile(o)
        except TypeError:
            return True

>       return src is None or not os.path.abspath(src).startswith(robot_path)
E       TypeError: startswith first arg must be str or a tuple of str, not WindowsPath

C:\Users\caden\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyfrc\tests\docstring_test.py:26: TypeError
```

### Operating System

Windows

### Installed Python Packages

```text
Package                   Version
------------------------- -----------
annotated-types           0.6.0
attrs                     23.2.0
bcrypt                    4.1.2
bokeh                     3.4.2
CacheControl              0.12.14
certifi                   2024.7.4
cffi                      1.16.0
charset-normalizer        3.3.2
colorama                  0.4.6
contourpy                 1.2.1
cryptography              41.0.7
easygui                   0.98.3
exceptiongroup            1.2.2
frc6343                   0.2
h11                       0.14.0
idna                      3.7
iniconfig                 2.0.0
Jinja2                    3.1.4
keyboard                  0.13.5
MarkupSafe                2.1.5
msgpack                   1.0.8
numpy                     2.1.2
opencv-python             4.9.0.80
outcome                   1.3.0.post0
packaging                 23.2
pandas                    2.2.2
paramiko                  3.4.0
phoenix6                  24.3.0
photonlibpy               2024.2.8
pillow                    10.4.0
Pint                      0.23
pip                       24.2
pluggy                    1.3.0
pycparser                 2.22
pydantic                  2.7.1
pydantic_core             2.18.2
pyfrc                     2024.0.1
PyNaCl                    1.5.0
pynetconsole              2.0.4
pyntcore                  2024.3.2.0
PySocks                   1.7.1
pytest                    7.4.4
pytest-reraise            2.1.2
python-dateutil           2.9.0.post0
pytz                      2024.1
PyYAML                    6.0.1
requests                  2.32.3
robotpy                   2024.3.2.1
robotpy-apriltag          2024.3.2.0
robotpy-cli               2024.0.0
robotpy-commands-v2       2024.3.1
robotpy-cscore            2024.3.2.0
robotpy-ctre              2024.1.3
robotpy-hal               2024.3.2.0
robotpy-halsim-ds-socket  2024.3.2.0
robotpy-halsim-gui        2024.3.2.0
robotpy-halsim-ws         2024.3.2.0
robotpy-installer         2024.2.0
robotpy-navx              2024.1.1
robotpy-pathplannerlib    2024.2.7
robotpy-playingwithfusion 2024.2.0
robotpy-rev               2024.2.4
robotpy-romi              2024.3.2.0
robotpy-wpilib-utilities  2024.0.0
robotpy-wpimath           2024.3.2.0
robotpy-wpinet            2024.3.2.0
robotpy-wpiutil           2024.3.2.0
robotpy-xrp               2024.3.2.0
selenium                  4.23.0
setuptools                69.0.3
six                       1.16.0
sniffio                   1.3.1
sortedcontainers          2.4.0
statbotics                2.0.3
tomli                     2.0.1
tornado                   6.4.1
trio                      0.26.0
trio-websocket            0.11.1
typing_extensions         4.9.0
tzdata                    2024.1
urllib3                   2.2.2
websocket-client          1.8.0
wpilib                    2024.3.2.0
wsproto                   1.2.0
xyzservices               2024.6.0
```

### Reproducible example code

```text
from pyfrc.tests.docstring_test import * # in tests/pyfrc_test.py
```

virtuald, 2024-10-16:
Weird. What version of Python?

PickleFace5, 2024-10-16:
3.12

virtuald, 2026-04-20:
pyfrc's testing functionality will be merged into mostrobotpy for 2027, and this test wasn't carried forward.