robotpy/pyfrc #239: Isolated mode fails with non-deterministic parameterization

virtuald, 2025-02-20:
There's probably a clever way to support this, but I'm not sure it's worth the time. Here's something that fails:

```python
def get_alliance_stations() -> list[str]:
    stations = (1, 2, 3)
    if "CI" in os.environ:  # pragma: no branch
        return [
            f"{alliance}{station}"
            for alliance in ("Blue", "Red")
            for station in stations
        ]
    else:  # pragma: no cover
        return [f"Blue{random.choice(stations)}", f"Red{random.choice(stations)}"]

@pytest.mark.parametrize("station", get_alliance_stations())
def test_fuzz(control: TestController, station: str) -> None:
    ...
```

virtuald, 2025-02-20:
An ugly workaround that a user can do for this situation is the following:

```python

def get_alliance_stations() -> list[str]:
    import json
    cached = os.environ.get("ALLIANCE_CACHE", None)
    if cached:
        return json.loads(cached)

    stations = (1, 2, 3)
    if "CI" in os.environ:  # pragma: no branch
        choices = [
            f"{alliance}{station}"
            for alliance in ("Blue", "Red")
            for station in stations
        ]
    else:  # pragma: no cover
        choices = [f"Blue{random.choice(stations)}", f"Red{random.choice(stations)}"]

    os.environ["ALLIANCE_CACHE"] = json.dumps(choices)
    return choices
```

auscompgeek, 2025-02-20:
pytest-xdist doesn't support this either, so if they don't have a solution I also don't think it's worth us spending the time.

closed 2026-04-20 by virtuald