robotpy/pyfrc #240: Isolated mode: only isolate tests that use the robot fixture

virtuald, 2025-02-21:
Some tests don't require isolation, we should detect those cases and run them normally to avoid the speed penalty.

auscompgeek, 2025-02-21:
We should probably stop autousing the robot fixture then, right? https://github.com/robotpy/pyfrc/blob/27fe979f7af9419c05c6c42c79f64d519781c089/pyfrc/test_support/pytest_plugin.py#L76

auscompgeek, 2025-03-01:
This is a bit of a killer for my team's code. Our unit tests each take noticeably longer to run with `--isolated`, doubling the total runtime of the tests (from ~14s to ~29s) on my M2 MBP.

etiennebeaulac, 2025-03-11:
Not directly related, but I think it would be a great idea to stop using the `robot` fixture automatically. My team has a number of tests which do not require the robot at all, but even for them, the overhead of creating and destroying the robot is still there. It also creates a number of problems with testing individual components, when you want to use some ports that are used by the robot.